eclipse-threadx/filex #13: _fx_fault_tolerant_apply_logs not updates fx_media_available_clusters of FAT logs when called after power down

smithBraun, 2024-01-09:
When __fx_fault_tolerant_apply_logs_ called in regular situation, the _fx_media_available_clusters_ is already updated in advanced, except of the FAT chain, and hence no need to update _fx_media_available_clusters_.
But when __fx_fault_tolerant_apply_logs_ is called after power down (in __fx_fault_tolerant_enable_), still just the FAT chain cleanup changes _fx_media_available_clusters_.
This may cause long time wasting for searching on non existing free cluster (when the FAT log captured cluster), or refusing to use available clusters because thinking there is no more (when the FAT logs released cluster).
I suggest that in __fx_fault_tolerant_apply_logs_ if fx_media_fault_tolerant_enabled is false (which means it was called from __fx_fault_tolerant_enable_), check for each FAT log if it changing the cluster state and if it is - update _fx_media_available_clusters_.

amgross, 2024-07-02:
Hi,
Same issue exist if operation failed, _fx_fault_tolerant_recover should revert fx_media_available_clusters state

amgross, 2024-07-02:
There are also multiple places (not related to fault tolerance) where flow can fail between updating fx_media_available_clusters to updating the FAT chain itself, hence I think maybe this update should be done inside fx_media_available_clusters itself.
This approach is bit complicated with fault tolerance, but it is the clean way to update when actually it changes.