eclipse-threadx/filex #52: How do you get the size of an open file? How do you copy a file?

111llll111, 2024-03-12:
Hi ，

How do you get the size of an open file? 
How do you copy a file?

I didn't find an api to get the size of a file. 
For example, if I want to copy a file and rename it to another file, what should I do?

Thanks！

rahmanih, 2024-03-12:
Hi @111llll111,
you can use the APIs:
```
UINT    fx_directory_first_full_entry_find(FX_MEDIA *media_ptr, CHAR *directory_name,
                                           UINT *attributes, ULONG *size, 
                                           UINT *year, UINT *month, UINT *day, 
                                           UINT *hour, UINT *minute, UINT *second);
```
``` 
UINT    fx_directory_first_full_entry_find(FX_MEDIA *media_ptr, CHAR *directory_name,
                                           UINT *attributes, ULONG *size, 
                                           UINT *year, UINT *month, UINT *day, 
                                           UINT *hour, UINT *minute, UINT *second);
```
regards
haithem.

amgross, 2024-03-12:
There is no straight way to copy file as far as I know. (To just rename there is API `fx_file_rename`)
To know file size you can open the file with  `fx_file_open` and then read the `fx_file_current_file_size` attribute from the `file_ptr`

111llll111, 2024-03-12:
The `FX_FILE `struct has the `fx_file_current_file_size  `so I can open two files at the same time, reading and writing at the same time is equivalent to copying.

closed 2024-03-12 by 111llll111