eclipse-threadx/filex #53: Write Performance on NAND Flash is very low when using Fault Tolerant Mode

sonmara, 2024-03-18:
I'm using a NAND-Flash (W25N01GV) in combination with a STM32H7B3 controller.
After enabling the FaultTolerantMode and writing data to to the flash, the write performance gets very low.
The W25N01GV has the following coarse specs:
	• FlashSize 125MByte
	• 2kB Page Size
	• 1024 Blocks with 64 pages
	• using quadSPI
On my controller I'm using threadX, fileX and levelX with the version 6.4.0. and the TX_TIMER_TICKS_PER_SECOND are set to 1000.
To compare different settings I did a write of 8MByte of data to the flash. The write is done by chunks of 512Byte.
```
for(int i= 0; i< 8192*2; ++i)
{
    fx_file_write(&_file, buffer, 512);
}
```
The settings of the fx_media_format was the following:
```
fx_media_format(&_media,
                          nandDriver,                         // Driver entry
                          nullptr,                                // Device info pointer
                          (UCHAR*)sectorBuffer,       // Media buffer pointer
                          2048,                                  // Media buffer size
                          "C",                                     // Volume Name
                          1,                                        // Number of FATs
                          32,                                      // Directory Entries
                          0,                                        // Hidden sectors
                          (1024 - 1) * 64,                   // Total sectors
                         2048,                                   // Sector size
                          1,                                        // Sectors per cluster
                          1,                                        // Heads
                          1);                                       // Sectors per track
```
### First Setp
Measuring write Speed **without** FaultTolerantMode enabled:
Without calling `fx_fault_tolerant_enable` the write speed is about 500kByte/sec.

### Second Step
Measuring write Speed **with** FaultTolerantMode enabled:
Compared to the first step, I only enabled the fault tolerant mode in the fx_user.h and called `fx_fault_tolerant_enable()`:
```
fx_fault_tolerant_enable(&_media
                          (VOID*)faultTolerantBuffer,          // faultTolerantBuffer 
                           4096);                                           // faultTolerantBuffer size
```
At the beginning of the 8MByte write process the write speed is 13kByte/sec.
But after short time, the write speed breaks down to 2kByte/sec.
I also played a bit with the possible defines in the fx_user.h and lx_user.h (e.g. LX_NOR_SECTOR_MAPPING_CACHE_SIZE) but nothing had an impact on the write speed.

### Questions
* Do I have to accept this low writing speed?
* Is the setting of fx_media_format correct?
* Is there a possibility to increase the write performance?
* What kind of flash (sector size, etc) would be the best for highest write performance?

rahmanih, 2024-05-09:
Hi @sonmara

Could you try to call the` lx_nand_flash_extended_cache_enable()` API after openeing the NAND flash instance.

regards
Haithem.

sonmara, 2024-05-16:
Hi @rahmanih,

in the levelx version in use this function is deprecated and looks like the following:

```
UINT  _lx_nand_flash_extended_cache_enable(LX_NAND_FLASH  *nand_flash, VOID *memory, ULONG size)
{

    LX_PARAMETER_NOT_USED(nand_flash);
    LX_PARAMETER_NOT_USED(memory);
    LX_PARAMETER_NOT_USED(size);
    
    /* Return not supported.  */
    return(LX_NOT_SUPPORTED);
}
```
I don't think its worth trying to call that one, sorry.