eclipse-threadx/filex #56: Standalone FileX platform support

Dharani3110, 2024-08-14:
Usecase: Nand flash filesytem for MCU board

is standalone Filex supported in Windows, Linux and Mac?

nflandin, 2024-09-21:
FileX manages FAT and exFAT filesystems, and these filesystems are supported by major computer operating systems. FileX is not meant to interact with computers. If you want to provide for a computer to access your filesystem, you need to have some sort of of driver (and maybe middleware like USBX).

vlang-intona, 2024-11-19:
FileX is not written in a very portable manner, but I got it to work on Linux. You have to workaround some issues like https://github.com/eclipse-threadx/filex/issues/64 or the utterly idiotic way how time handling works. You can start with the generic port.

But why the hell would you use an awful, inefficient, and limited filesystem such as FAT for something internal?

fdesbiens, 2024-11-19:
> FileX manages FAT and exFAT filesystems, and these filesystems are supported by major computer operating systems. 

@nflandin Microsoft removed exFAT support when they contributed the code to the Eclipse Foundation. exFAT is still covered by at least one patent. FileX still supports FAT 12, 16, and 32.

closed 2024-11-19 by Dharani3110

fdesbiens, 2024-11-19:
@vlang-intona Commercial implementations from companies who licensed the patent could exist. I am not aware of any.

vlang-intona, 2024-11-19:
@fdesbiens I assume Microsoft didn't just trash their exFAT support completely. But azure-rtos was made read-only. Maybe it was renamed again? If it exists, I can't find it.

nflandin, 2024-11-19:
> @nflandin Microsoft removed exFAT support when they contributed the code to the Eclipse Foundation. exFAT is still covered by at least one patent. FileX still supports FAT 12, 16, and 32.

When was exFAT support removed? Commit #8c216a298f9252bd4e3af9f68670646f5aecdbc1 "Added exFAT licensing blurb", and I understand that exFAT may not be currently worked on, but it doesn't appear to have been removed (as far as I can tell).

fdesbiens, 2024-11-19:
> When was exFAT support removed? Commit #8c216a298f9252bd4e3af9f68670646f5aecdbc1 "Added exFAT licensing blurb", and I understand that exFAT may not be currently worked on, but it doesn't appear to [...]

@nflandin The commit you refer to exists in the Azure RTOS repository. If you browse the Eclipse FileX commit history, you will notice the repository only contains the 6.4.1 release, whereas we have the full 6.x series history for all other components. exFAT is definitely not part of the Eclipse ThreadX codebase.