eclipse-threadx/filex #60: If I port Filex to another RTOS without TheadX, do I need to maintain read/write protection in multithreading?

XingYeKM, 2024-09-12:
I noticed that read and write protection in Filex is done with the macro definition FX_PROTECT, which uses ThreadX.
[image]

nflandin, 2024-09-20:
In short, yes.

Attempting to access a filesystem from one thread while another thread is modifying it can lead to many problems. If you have more than one thread modifying and/or accessing any filesystem, you need to have some way of protecting against potential problems. This can take the form of the FX_PROTECT/FX_UNPROTECT mutex macros, or an infinite number of other possibilities.

XingYeKM, 2024-09-23:
Thanks @nflandin 
I know what to do, okay

closed 2024-09-23 by XingYeKM