eclipse-threadx/filex #63: About  read/write performance degradation (recovered after power-on but continued to decline)

XingYeKM, 2024-11-11:
Enable the fault tolerance module when using FileX, and then define FX_SINGLE_THREAD. Multithreaded protection FX_PROTECT and FX_UNPROTECT are maintained by my own RTOS. In the case of continuous read and write, FileX's read and write performance becomes worse and worse, and it even takes 180 seconds to create a file or directory! However, when I powered it on again, the read and write performance returned to the original speed, and then the speed continued to decline.

The structure of the file system and the contents of the files are fine
The disk I use is a 20GB SSD

What causes this?

fdesbiens, 2024-11-11:
Thank you for reaching out @XingYeKM. It's hard to tell what's happening without additional details. 

Please answer the following questions to help us understand this issue:
- What kind of partition is this? FAT12, 16 or 32?
- What size is the partition?
- What kind of storage device are you using?
- If using flash, Is LevelX enabled?
- What hardware platform are you using? Did you try your code on a different platform or storage device?
- Did you try running your code on a Linux environment? If so, did you see the same behaviour? 
-  Did you instrument your application? What specific API calls are taking more time to execute?

Also, which specific version of FileX is involved?

Thanks in advance for your answers.

fdesbiens, 2024-11-21:
We cannot help if you do not provide additional information, @XingYeKM.

Also, did you try running the code on top of ThreadX rather than using your own RTOS?

fdesbiens, 2024-12-05:
@XingYeKM If you do not provide additional details soon, I will close this issue.

fdesbiens, 2024-12-12:
Closing this issue for now. Please feel free to reopen @XingYeKM.