eclipse-threadx/filex #69: Race condition leads to files in a wrong directory

sschneideribs, 2025-05-06:
Greetings,

We are using ThreadX in combination with FileX, and we are making FileX calls from multiple threads, as FileX is advertised as being thread-safe when used with ThreadX. Recently, we encountered an issue where a call to `fx_file_create` created a file in the root directory instead of the specified subdirectory, even though the subdirectory did exist.

After an extensive debugging session, we discovered that a race condition between threads was overwriting the path name, leading to the behavior described. Here's how the issue can be reproduced:

1. Create a subdirectory, for example: /Alarms/
2. Create a file within that subdirectory: /Alarms/0001_alarm.json
3. While `fx_file_create` is searching for an existing file name inside `_fx_directory_search`, call another FileX function that modifies the media directory name, such as `fx_directory_local_path_set`.
4. As a result, the file 0001_alarm.json ends up being created in the root directory instead of /Alarms/.

The race condition is easier to trigger when many files exist in the same directory, since the search takes longer, increasing the likelihood of the conflict. The root of the problem appears to be that `FX_PROTECT` is called too late. It needs to be invoked at the very beginning—before the media directory entry name is [nulled](https://github.com/eclipse-threadx/filex/blob/master/common/src/fx_directory_local_path_set.c#L113).

I have observed that several other functions may be susceptible to the same behavior. As a temporary workaround, we have wrapped all FileX function calls that are not properly protected with an additional mutex.

sschneideribs, 2025-05-22:
@fdesbiens Any Updates on this topic?

IlkkaHokkanen, 2025-10-17:
Struggling with same/similar issue and implementation in _fx_directory_local_path_set() looks suspicious.

There seems to be usage of "media_ptr -> fx_media_name_buffer" which is not protected and function is not re-entrant. 
```
/* Setup pointer to a name buffer.  */
    dir_entry.fx_dir_entry_name =  media_ptr -> fx_media_name_buffer + FX_MAX_LONG_NAME_LEN;
```

For me it seems there should be FX_PROTECT before using "fx_media_name_buffer"

sschneideribs, 2025-10-17:
@billlamiework The mutex works as expected but it is called too late.

@IlkkaHokkanen You are right, this call needs to be proteced by the FileX mutex, which is not the case. Therefore wrapping all function calls and protect them with an own mutex seems to be the only solution for us right now.